christianrauch/msp #6: Motors keep spinning

Anderson321, 2017-05-17:
Hi! Thanks for the last quick response. I am now trying to run fcu_motors and I keep getting a bunch of CRC not matching errors. I am using a naze32 with cleanflight 1.11 firmware installed on it. The receiver option right now is RX_MSP and pretty much every time I try to run any of the examples I get something like "CRC not matching: Message 101, expected CRC 123, received CRC 168"

Also, when I try to run fcu_motors, the motors start, but then they won't stop even when the program has stopped running. I dont have Dynbalance included, is that necessary? Without Dynbalance will I not be able to send correct commands?

Thanks :)

Anderson321, 2017-05-17:
Ok i fixed it. When I put a std::this_thread::sleep_for(std::chrono::miliseconds(5000); after the stop command it worked. But anyways I still get a lot of CRC check sum errors. thoughts?

christianrauch, 2017-05-18:
CRC errors basically indicate that the message is corrupted. They are usually ignored in this case.
This is mostly caused by faults on the serial line, e.g. cause by electromagnetic interference or just bad wiring.
What kind of serial connection are you using? Are you using the USB connection, respectively the  USB-UART adapter on the naze32?

Anderson321, 2017-05-18:
I am just using a usb to microusb connection from rpi to naze32

christianrauch, 2017-05-18:
I am using a naze32 rev6 with betaflight 3.1.7 connected over USB at 500000 bit/s. I can run `fcu_test` and `fcu_motors` without CRC errors and the motors start and stop as expected.
Do you also experience connection errors when using the Cleanflight or Baseflight Configurator?

Anderson321, 2017-05-18:
Yeah sometimes i cant make a connection with cleanflight. I'll have to replug it a couple times. Is this maybe it?

Anderson321, 2017-05-18:
I'm just using like an android micro usb

christianrauch, 2017-05-18:
I guess it is related to the serial connection. You could try to use a different cable or USB port (or a different computer if you can).

christianrauch, 2017-05-20:
I've tested with Cleanflight version 1.11 and I also receive CRC errors. When using a recent version, e.g. 2.0.2, these CRC errors do not occur any more.
I simply recommend that you upgrade your cleanflight firmware to the latest stable release :-)

closed 2017-05-26 by Anderson321