christianrauch/msp #7: Dynbalance

Anderson321, 2017-05-18:
Hey so I can't send RC commands but I can set motor values because I dont have DYNBALANCE included. Since I am using cleanfliht and not MultiWii code, where should i put "#define DYNBALANCE" in Multiwii's config.h?

thanks

Anderson321, 2017-05-18:
Where do I *comment* out DYNBALANCE

correction

christianrauch, 2017-05-18:
Have you enabled the `RX_MSP` feature? When you type `feature` in the cleanflight/baseflight CLI, it should print a list that contains `RX_MSP`. Otherwise you need to select the receiver type (PPM, PWM, SBUS, MSP) in the configuration tab.

Anderson321, 2017-05-18:
Right now, I want to configure cleanflight for RX_PPM, control it with a RC and then have the raspberry pi take over and in your code in fcu_arm_test.cpp:
You enable RX_MSP automatically so that shouldn't be a problem.
    // on cleanflight, we need to enable the "RX_MSP" feature
    if(fcu.isFirmwareCleanflight()) {
        if(fcu.enableRxMSP()==1) {
            std::cout<<"RX_MSP enabled, restart"<<std::endl;
            feature_changed = true;
            goto start;
        }

I am able to setMotors perfectly fine but I can't send RC commands, which I think is the DYNBALANCE thing, right?

Anderson321, 2017-05-18:
Ok so update: I put in

if(fcu.isFirmwareCleanflight()) {
if(fcu.enableRxMSP()==1) {
std::cout<<"RX_MSP enabled, restart"<<std::endl;
feature_changed = true;
goto start;
}

to enable RX_MSP and then after that I am doing:

fcu.setRc(1500,1500,1500,1500,1500,0,0,0);
std::this_thread::sleep_for(std::chrono::milliseconds(3000));

My AUX 1 channel after 1500 is to arm the FC and then the 4 other channels should be throttle pitch yaw roll etc. Nothing is happening though. Thanks for your time

christianrauch, 2017-05-18:
`#define DYNBALANCE` is related to the original MultiWii firmware for Arduino / 8bit AVR microcontroller. I would recommend to select `RX_MSP` in the GUI or CLI beforehand since it requires to reboot the FC to apply the new settings.

Can you confirm that you have `RX_MSP` enabled and that the virtual control sticks in the Receiver tab of the Cleanflight GUI allow you to change the channels and arm / disarm the FC?

Anderson321, 2017-05-18:
I have RX_MSP enabled but the virtual sticks aren't working, I even have my AUX 1 at 1500 (which is arming it)... Scratch that, the motors started after awhile with the virtual control sticks... but now doesn't start up again. It's very inconsistent.

Also, I went into FCU.arm() and I changed it to this so it would work for my arming configuration:

const uint16_t aux1= arm ? 1500 : 1000;

return setRC(1000, 1000, 1000, 1000, aux1, 1000, 1000, 1000);
I've debugged it and it's getting caught in an infinite loop in FCU.arm_block()  where it goes into arm(), trying to setRC to (1000, 1000, 1000, 1000, aux1, 1000, 1000, 1000) where aux1 is 1500.

christianrauch, 2017-05-18:
Can you check if arming works if you switch the channel map in the Receiver tab? It might be possible that only one of them (either `AETR1234` or `TAER1234`) is working. It should always be possible to arm via the stick combo.

Anderson321, 2017-05-18:
Yeah, it's very inconsistent for each stick combo...

christianrauch, 2017-05-18:
Is `FlightController::arm()` arming when you switch to `AETR1234`?

Anderson321, 2017-05-19:
It doesn't arm in either configuration.

Anderson321, 2017-05-19:
I have debugged it and during arm it enters the FlightController::setRC command with the parameter (1000, 1000, 1000, 1000, 1500, 1000, 1000, 1000) but nothing happens. RX_MSP is enabled as well

christianrauch, 2017-05-19:
Can you provide a link from the [Cleanflight release page](https://github.com/cleanflight/cleanflight/releases) to the exact Cleanflight version (cleanflight_*_NAZE.hex) that you are using? Can you also upload a dump of your configuration (type `dump` in the CLI)?

I will try to reproduce your setup and see if I encounter the same issues you are.

Anderson321, 2017-05-19:
https://github.com/cleanflight/cleanflight/tree/v1.11.0/docs I'm using 1.11 

# version
# Cleanflight/NAZE 1.11.0 Nov 28 2015 / 15:30:06 (0c82a7b)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature FAILSAFE
feature TELEMETRY
feature RX_MSP
feature ONESHOT125

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
[... 251 lines cut ...]
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set dterm_cut_hz = 0
set pterm_cut_hz = 0
set gyro_cut_hz = 0

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

Thank you for your help I appreciate it

Anderson321, 2017-05-19:
I am using a rev5 naze btw

christianrauch, 2017-05-20:
I can confirm that arming is working with Cleanflight version 1.14.2.
Beginning with version 2.0, `MSP_SET_RAW_RC` messages are ignored for some targets, e.g. for the NAZE. You will recognise this, if the `MSP_RC` message is empty (e.g. run `fcu_test` and check "Rc channels" is 0).

The solution to this problem is to either stick with version 1.14.2 or to recompile your own Cleanflight version 2. In the latter case, you need to add `#define USE_RX_MSP` to the `target.h`.

For example, I am maintaining my own betaflight version for NAZE32 rev6 targets that removes sensors that are not present and adds `USE_RX_MSP`: https://github.com/christianrauch/betaflight/tree/naze_rev6_full. You can have a look at how I patched file `src/main/target/NAZE/target.h` and apply similar changes to your target.

Anderson321, 2017-05-20:
Thank you! I will try this as soon as possible and let you know if it worked. Would your betaflight version work on rev5 or would i need rev6?

Anderson321, 2017-05-22:
So I switched to 1.14.2... I still cannot arm. I will try to arm and then I run fcu_test and none of the RC channels change. Also in cleanflight configurator I made it so all values of aux1 would arm the drone (and I made it so motors would spin when armed), but when i switch to RX MSP the motors don't spin. It's like AUX1 when switched over to MSP isn't arming the drone. Do you have any thoughts?

christianrauch, 2017-05-22:
`fcu_test` will not change the channels values, it only reads messages from the FC.

Have you updated to the newest version of the library? I recently fixed a bug related to the RX channel mapping.

Anderson321, 2017-05-22:
What I mean is that I try to arm it (I set aux to 1500) and then do fcu_test to see if the RC value changed after the fact. But the RC values are the same as before.  I updated to the newest version of the library.

Also. I tried arming it this morning, it armed. However, I'll try to arm it again and then it will get stuck in the arm_block() function.

Anderson321, 2017-05-22:
I will also run msp_read_test and my value of aux1 is 1500, which is confusing to me because that is in the range of being armed according to my configuration, but it doesn't think it is armed.

Anderson321, 2017-05-22:
I'm going to try flashing your betaflight firmware onto my Naze. I'm not sure how to flash your version locally. How do I do this?

christianrauch, 2017-05-25:
It is important to note that when controlling the FC via `MSP_SET_RAW_RC` messages, the FC is affected by the same failsafe rules that are applied to regular controls, e.g. via a PPM or PWM signal. That means that you need to provide new values for the channels at regular intervalls, e.g. at least every 1 second. If you do not provide new values within the failsafe guard time, the FC will go into failsafe mode and it will set default values for these channels. I guess, these are the values that you see when reading the `MSP_RC` message.

I recommend to write your own program that continuously provides channel values for the FC, e.g. from a joystick or keyboard. You can also have a look at ROS and the MultiWii ROS node (https://github.com/christianrauch/ros-multiwii) and use another ROS node to continuously provide commands.

If you are able to set channel values via the `MSP_SET_RAW_RC` message on cleanflight 1.14.2 please close this issue.

Anderson321, 2017-05-26:
I got it to arm! For some reason, I wasn't able to arm through aux 1, but I can arm by setting yaw to 2000 and throttle 1000, and the other two to 1500. Yeah I have noticed that I need to continuously update the channels. I was trying to just setRc and then sleeping for like 5 seconds but it stopped before the sleep was over. Is 1 second a good rule of thumb? Thanks

closed 2017-05-26 by Anderson321