christianrauch/msp #9: Issues with tests

spacehamster, 2017-11-21:
When I run the tests in the examples, fcu_custom_type and fcu_test crash when `FlightController fcu` is deconstructed, and client_read_test stalls when `client.stop()` is called.

With fcu_custom_type and fcu_test, I suspect that they crash because `PeriodicTimer` in `SubscriptionBase` is deleted without stop being called. Calling stop before deleting seems to block until the next timer tick, so maby a timed mutex could stop the timer faster?

As for client_read_test, it seems to be stuck on `asio::read_until(port, buffer, "$M");` at `Client::processOneMessage`, so i'm not quite sure whats going on there.

By the way, I also needed to make a few changes to get it to compile on windows, changing `uint` to `unsigned int`  and marking the functions in `deserialise.hpp` as static.

spacehamster, 2017-11-22:
Created pull request, still not sure what is wrong with client_read_test

christianrauch, 2017-11-23:
Thanks for your contribution.
In the end, I replaced all `uint` with `size_t` to be more standard compliant. Can you checkout branch `win_build`, which contains these changes, and check if it still works for you?

spacehamster, 2017-11-23:
branch `win_build` still works, although `client_read_test` still blocks. 
I'm not very familiar with asio, but I suspect it might because the serial port isn't closed before the `Client` thread is joined, so `asio::read_until(port, buffer, "$M");` never returns.

christianrauch, 2017-11-24:
`asio::read_until` blocks and that is why `thread` can never join at the end of `client_read_test`, e.g. after all responses have been received. In `client_async_test` it is working since the background thread is still sending (and receiving) messages, which unblock `asio::read_until`.

I pushed an update that detaches from the thread instead of waiting for the result. This should prevent blocking of the program.

spacehamster, 2017-11-24:
I don't think detaching the thread is the thing to do, because now fcu_test crashes on exit because the `Client` thread keeps running even after `Client` has been destroyed, and tries to access deallocated objects.

christianrauch, 2017-11-24:
Thanks for the fixes in PR #12. I am wondering if there are more of these kind of issues (improper deallocation of members at destruction of objects) and if there is a good way of systematically discovering them with tests.

spacehamster, 2017-11-25:
I don't have much experience writing robust code,  but it is my intuition is that tests won't uncover unexpected edge cases, because they are never thought to test for (the unknown unknowns). I don't have any experience with it, but perhaps a tool like valgrind could help? Another idea might be to add in asserts to expose when basic assumptions are violated?

closed 2017-12-08 by christianrauch