christianrauch/msp #15: Issue with msp.request_block(box)

naifedicer, 2018-03-13:
the msp.request_block(box) method is currently "not supported"

christianrauch, 2018-03-13:
What do you mean by that? The `MSP::request_block()` method is still available:
https://github.com/christianrauch/msp/blob/33adb2be5c4566877b674063f29363b3236b6671/inc/msp/MSP.hpp#L120 and it is used in one of the examples: https://github.com/christianrauch/msp/blob/33adb2be5c4566877b674063f29363b3236b6671/examples/msp_read_test.cpp#L93

Can you provide more details? E.g. which code are you trying to run and what is the flight controller firmware + version that you are using?

naifedicer, 2018-03-14:
I know it is used in one of the examples. But it keeps returning 113 not
supported​. I am using a spracingf3 flight controller.

christianrauch, 2018-03-14:
Which firmware (and which version of it) are you using? Are you sure that your firmware supports this message? E.g. in the last stable release of betaflight (version 3.3.0), `MSP_BOX` is declared as deprecated:
 https://github.com/betaflight/betaflight/blob/177472b4fc135afc329256f1d692f3644742594a/src/main/interface/msp_protocol.h#L257.
The low-level API (classes `MSP` and `Client`) does not consider the type or version of the firmware, it just communicates with the flight controller and sends/receives the raw messages. For a high-level access to the flight controller, consider using the `FlightController` class, which provides more functionality than just sending the raw messages..

christianrauch, 2018-03-27:
Has your issue been solved?

closed 2018-04-08 by christianrauch