christianrauch/msp #16: Error Compiling on RPi Arch

emwjacobson, 2018-03-23:
After cloning repo I entered the folder and ran the `mkdir build && cd build && cmake ..&& make -j` and got this response:
https://pastebin.com/C0sHTZ1y

I installed [this](https://www.archlinux.org/packages/extra/x86_64/asio/) version of asio, and have devtools installed.

Im not sure what to do to resolve this, any help would be appreciated.

christianrauch, 2018-03-23:
The C++11 standard types `function`, `bind`, `placeholders` are missing.This might simply be caused by a missing include of  `<functional>`, since all three types are defined there. I am using gcc 5.4.0, but do not get this error. There might be a difference in how gcc 5and 7 handle the order of includes.

Can you simply try to compile with gcc 5.4? Or alternatively add `#include <functional>` to the header and send a PR if this solves the problem.

christianrauch, 2018-03-23:
I just added this header anyway, since I got the same problem under Ubuntu 18.04. Just update the master branch and build again.

emwjacobson, 2018-03-23:
https://github.com/christianrauch/msp/commit/e0136692e0ecbaf8faf340fa3e58eae1e196e3e7 has fixed my problem, thanks!

closed 2018-03-23 by christianrauch