christianrauch/msp #21: Incorrect GPS longitude value??

mwingerson, 2018-05-22:
I was trying to get the current GPS of my vehicle streamed to my laptop and when I look at my longitude, I am getting about 312 degrees while I should be getting -117 degrees. I am fairly sure that I am just missing something simple.

Thanks!
marsall

christian-rauch, 2018-05-22:
I only added the message definition from the MSP message overview, but I never used it. It might be an error during the deserialisation of the payload.
Have you tried to read the GPS values with one of the official GUI programs?
Which firmware afe you using? It might also be that the message definition changed between thd original MultiWii and cleanflight/betaflight.

christianrauch, 2018-05-22:
I only added the message definition from the MSP message overview, but I never used it. It might be an error during the deserialisation of the payload.

Have you tried to read the GPS values with one of the official GUI programs?

Which firmware are you using? It might also be that the message definition changed between thd original MultiWii and cleanflight/betaflight.

mwingerson, 2018-06-01:
Sigh....it was just that I needed to cast the long and lat to a signed value.

Thanks!

christianrauch, 2018-06-02:
Are you referring to the `MSP_RAW_GPS`? I am wondering why you need to cast the value to a signed. The `lon` and `lat` are given in 1/10000000 degree. Simply casting these uint32 to int32 will give you wrong values.

closed 2018-06-13 by christianrauch

mwingerson, 2018-06-13:
Sorry Christian I lost this conversation in my emails :/

For the firmware, I am using INAV/MATEKF405 1.9.0

The protocol documentation showed that the uint32_t values for lat and lon are given in 1/10000000 degrees but the lat and lon are not defined as a int32_t.  They are defined as uint32_t which confused me until I realized the value should be signed.

But I figured it out. Thanks!