christianrauch/msp #23: Unusual connection issues with Quadrino Nano

primetoxinz, 2018-06-01:
I am attempting to use msp with a Quadrino Nano with MultiWii v2.4
When attempting to make an initial connection confirmation request, if the request is made too soon the board never recieves the command and thus never replies.

The basic request wait from the test suite:
```
//MSP initialization
 if(msp.request_wait(ident, 1000))
//Other code
```
In my case, if the request is made any sooner than 1s (1000ms) it never replies with the identification as if it was never even seen in the first place. 
I am not entirely sure what the cause of this is, as I would expect that the ident request would simply stay in the buffer until it could be processed and eventually reply even if sent too soon, but that does not seem to the case.
If you have any ideas as to what could cause this it would be very helpful.

christianrauch, 2018-06-01:
This is a known problem. The Arduino boards with onboard USB-serial connection reset when a serial connection is established. You therefore need to wait until the board is ready to receive messages. Otherwise, messages send to the FC are lost and you will never receive a reply. But there are methods with a timeout parameter in the API to not block in these cases. In such a case you need to resent you request with a timeout a couple of times, until the board responds with a reply.

primetoxinz, 2018-06-04:
Right, thank you for the information. I did not know the Arduino had this problem. 
As for the timeout for the requests, I notice that FlightController has this line
```
void FlightController::initialise() {
    // wait for connection to be established
while(client.request(ident, 0.5)==-1);
...
```
Though the 0.5 seconds is not quite enough for the arduino to reset.
Could you possibly instead replace it with a call to `FlightController::waitForConnection` so that it can be overriden in subclasses or something similar if you prefer?

christianrauch, 2018-06-05:
From your other issue report (https://github.com/christianrauch/msp/issues/22) I see that you are able to run the `msp_read_test` example program. The important part in this example is `msp.request_wait(ident, 10)`, which sends requests until a response is received. The parameter `wait_ms` here is for the delay between writing and reading from the serial port, it's not the time required for the initialisation of the Arduino. However, `request_wait` is used for send requests until the Arduino is initialised and responds.

If you are using the `FlightController` (instead of `MSP`), sending and receiving is done asynchronous and does not block. E.g. if using `FlightController` in the same way used in the `fcu_test` example program, you do not need to explicitly wait for initialisation of the FC or set a delay between writing and reading from the serial port.
Does the `fcu_test` work for your Quadrino Nano? It works for my Arduino Nano.

closed 2018-08-03 by primetoxinz