christianrauch/msp #30: Can't compile version 3.0.0 on Windows

A-KL, 2019-03-13:
Hi,

I'm trying to compile version 3.0.0 on Windows but keep getting same errors on every machine I tried. Here is the full console log from Developer Command Proment for VS 2017:
```
C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional>cd C:\GitHub\MSP\build

C:\GitHub\MSP\build>cmake -DASIO_HEADER_PATH=C:\GitHub\asio-1.12.2\include ..
-- Building for: Visual Studio 15 2017
-- Selecting Windows SDK version 10.0.17134.0 to target Windows 10.0.17763.
-- The C compiler identification is MSVC 19.16.27027.1
-- The CXX compiler identification is MSVC 19.16.27027.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: C:/GitHub/MSP/build

C:\GitHub\MSP\build>cmake --build .
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  CMake does not need to re-run because C:/GitHub/MSP/build/CMakeFiles/generate.stamp is up-to-date.
  Building Custom Rule C:/GitHub/MSP/CMakeLists.txt
  CMake does not need to re-run because C:/GitHub/MSP/build/CMakeFiles/generate.stamp is up-to-date.
  Client.cpp
  Please define _WIN32_WINNT or _WIN32_WINDOWS appropriately. For example:
  - add -D_WIN32_WINNT=0x0501 to the compiler command line; or
  - add _WIN32_WINNT=0x0501 to your project's Preprocessor Definitions.
  Assuming _WIN32_WINNT=0x0501 (i.e. Windows XP target).
c:\github\msp\inc\msp\ByteVector.hpp(426): warning C4146: unary minus operator applied to unsigned type, result still unsigned [C:\GitHub\MSP\build\mspclient.vcxproj]
C:\GitHub\MSP\src\Client.cpp(10): error C2280: 'std::atomic_flag::atomic_flag(const std::atomic_flag &)': attempting to reference a deleted function [C:\GitHub\MSP\build\mspclient.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.16.27023\include\atomic(169): note: see declaration of 'std::atomic_flag::atomic_flag'
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.16.27023\include\atomic(169): note: 'std::atomic_flag::atomic_flag(const std::atomic_flag &)': function was explicitly deleted
  PeriodicTimer.cpp
C:\GitHub\MSP\src\PeriodicTimer.cpp(8): error C2280: 'std::atomic_flag::atomic_flag(const std::atomic_flag &)': attempting to reference a deleted function [C:\GitHub\MSP\build\mspclient.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.16.27023\include\atomic(169): note: see declaration of 'std::atomic_flag::atomic_flag'
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.16.27023\include\atomic(169): note: 'std::atomic_flag::atomic_flag(const std::atomic_flag &)': function was explicitly deleted
  Generating Code...
```

Is it somethig related to undefined `_WIN32_WINNT`?

christianrauch, 2019-03-13:
The issue is related to the deleted `atomic_flag` copy constructor:
> `error C2280: 'std::atomic_flag::atomic_flag(const std::atomic_flag &)': attempting to reference a deleted function`

This is probably a difference in compiler settings.
The Windows build successfully compiles for the CI setup: https://travis-ci.com/christianrauch/msp/jobs/175292197

christianrauch, 2019-03-15:
I just realised that the CI changes were on master but not released in `3.0.0`.
Can you test again on version 3.1.0?

christianrauch, 2019-03-18:
I assume this issue is solved in version 3.1. I will close this for now. Please reopen, if the issue still persists.

closed 2019-03-18 by christianrauch

A-KL, 2019-03-30:
Sorry, for the delay. Yes, thanks @christianrauch I just tested and it builds!