christianrauch/msp #40: g++ failing after linking with -lmsp_fcu -lmspclient.

pollito5, 2020-02-28:
I ran sudo make install after building the project to install the dependencies but g++ fails to link. I am adding the path to the libraries with -L/usr/local/lib, and compiling with -I/usr/local/include

christianrauch, 2020-02-28:
How is the linking related to the installation? The linking errors should already appear before installing. The install stage only copies files to their respective location and configures the pkg-config file.
Can you paste the linking errors that you get and show during which stage?

I also assumed that the include and library path should be provided by the package-config file. Can you paste your installed pkg-config file here?

pollito5, 2020-02-29:
Here is what my pkg-config file says:

Name: msp
Description: 
Version: 
Requires: 
prefix=/usr/local
includedir=${prefix}/include/
libdir=${prefix}/lib
Libs: -L${libdir} -lmsp_fcu -lmspclient
Cflags: -I${includedir}

This is occurring after installation. I'm trying to build a program using the header #include <msp/FlightController.hpp>. This is what I have attempted:

- Building was successful when #include <msp/FlightController.hpp> was added to main.cpp and compiled with: NOTE this was done with a single file.
g++ main.cpp -o main -L/usr/local/lib -lmsp_fcu -lmspclient 

- When adding #include <msp/FlightController.hpp> to a different to a header FCU.hpp defining class FCU and having fcu::FlightController be a member of the class. When I compile main.cpp and FCU.hpp/.cpp using a makefile:

main.o: main.cpp
  $(CC) $(CFLAGS) -c main.cpp -o $@
FCU.o: FCU.cpp
  $(CC) $(CFLAGS) -c FCU.cpp -o $@

main: main.o FCU.o
  $(CC) $(CFLAGS) -pthread -std=c++17 -o main main.o FCU.o -L/usr/local/lib -lmsp_fcu -lmspclient 

The following linking error was outputted.

FCU.o: In function `std::ostream& operator<< <unsigned char>(std::ostream&, msp::Value<unsigned char> const&)':
/usr/local/include/msp/Value.hpp:77: multiple definition of `std::ostream& operator<< <unsigned char>(std::ostream&, msp::Value<unsigned char> const&)'
fcuTest.o:/usr/local/include/msp/Value.hpp:77: first defined here
FCU.o: In function `std::ostream& operator<< <signed char>(std::ostream&, msp::Value<signed char> const&)':
/usr/local/include/msp/Value.hpp:86: multiple definition of `std::ostream& operator<< <signed char>(std::ostream&, msp::Value<signed char> const&)'
fcuTest.o:/usr/local/include/msp/Value.hpp:86: first defined here
FCU.o: In function `operator<<(std::ostream&, msp::ByteVector const&)':
/usr/local/include/msp/ByteVector.hpp:456: multiple definition of `operator<<(std::ostream&, msp::ByteVector const&)'
fcuTest.o:/usr/local/include/msp/ByteVector.hpp:456: first defined here
FCU.o: In function `msp::firmwareVariantToString[abi:cxx11](msp::FirmwareVariant)':
/usr/local/include/msp/FirmwareVariants.hpp:36: multiple definition of `msp::firmwareVariantToString[abi:cxx11](msp::FirmwareVariant)'
fcuTest.o:/usr/local/include/msp/FirmwareVariants.hpp:36: first defined here
FCU.o: In function `operator<<(std::ostream&, msp::ID const&)':
/usr/local/include/msp/Message.hpp:73: multiple definition of `operator<<(std::ostream&, msp::ID const&)'
fcuTest.o:/usr/local/include/msp/Message.hpp:73: first defined here
FCU.o: In function `operator<<(std::ostream&, msp::Message const&)':
/usr/local/include/msp/Message.hpp:78: multiple definition of `operator<<(std::ostream&, msp::Message const&)'
fcuTest.o:/usr/local/include/msp/Message.hpp:78: first defined here
FCU.o: In function `msp::armingFlagToString[abi:cxx11](unsigned int)':
/usr/local/include/msp/msp_msg.hpp:264: multiple definition of `msp::armingFlagToString[abi:cxx11](unsigned int)'
fcuTest.o:/usr/local/include/msp/msp_msg.hpp:264: first defined here
FCU.o: In function `operator<<(std::ostream&, msp::msg::ImuSI const&)':
/usr/local/include/msp/msp_msg.hpp:5499: multiple definition of `operator<<(std::ostream&, msp::msg::ImuSI const&)'
fcuTest.o:/usr/local/include/msp/msp_msg.hpp:5499: first defined here
collect2: error: ld returned 1 exit status
Makefile:64: recipe for target failed

christianrauch, 2020-02-29:
It is still not clear to me what the structure of your setup is. It sounds like you are including some header files multiple times.

Can you share a minimal example that produces this issue? I am still not sure if this is an issue of the map library or a generic linking issue.

Did you try to use cmake, as in the example https://github.com/christianrauch/multiwii_ros2?

pollito5, 2020-03-02:
Here is my setup I tested it to see if the linking error persisted and it did. 

I appreciate that your taking a look at this. 

[msp_github.zip](https://github.com/christianrauch/msp/files/4273215/msp_github.zip)

christianrauch, 2020-03-02:
The Makefile runs without problems in my setup (Ubuntu 18.04). I installed the library as Debian package using [`checkinstall`](https://wiki.debian.org/CheckInstall).

The produced `test` executable links against `libmsp_fcu.so`:
```
$ lddtree test 
test => ./test (interpreter => /lib64/ld-linux-x86-64.so.2)
    libmsp_fcu.so => /usr/local/lib/libmsp_fcu.so
        libmspclient.so => /usr/local/lib/libmspclient.so
            ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2
    libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6
    libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1
    libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6
```

pollito5, 2020-03-04:
I reinstalled the project using check install and am still getting the same linking error. Here is the output from checkinstall:

This package will be built according to these values: 

0 -  Maintainer: [ root@kevin-MacBookPro ]
1 -  Summary: [ msp.pc ]
2 -  Name:    [ build ]
3 -  Version: [ 20200303 ]
4 -  Release: [ 1 ]
5 -  License: [ GPL ]
6 -  Group:   [ checkinstall ]
7 -  Architecture: [ amd64 ]
8 -  Source location: [ build ]
9 -  Alternate source location: [  ]
10 - Requires: [  ]
11 - Provides: [ build ]
12 - Conflicts: [  ]
13 - Replaces: [  ]

Enter a number to change any of them or press ENTER to continue: 

Installing with make install...

========================= Installation results ===========================
[ 17%] Built target mspclient
[ 29%] Built target client_async_test
[ 41%] Built target msp_fcu
[ 52%] Built target fcu_test
[ 64%] Built target client_read_test
[ 76%] Built target fcu_arm
[ 88%] Built target fcu_custom_type
[100%] Built target fcu_motors
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/local/lib/libmsp_fcu.so
-- Up-to-date: /usr/local/lib/libmspclient.so
-- Up-to-date: /usr/local/include//msp
-- Up-to-date: /usr/local/include//msp/FlightMode.hpp
-- Up-to-date: /usr/local/include//msp/FirmwareVariants.hpp
-- Up-to-date: /usr/local/include//msp/FlightController.hpp
-- Up-to-date: /usr/local/include//msp/Message.hpp
-- Up-to-date: /usr/local/include//msp/PeriodicTimer.hpp
-- Up-to-date: /usr/local/include//msp/Client.hpp
-- Up-to-date: /usr/local/include//msp/Value.hpp
-- Up-to-date: /usr/local/include//msp/Subscription.hpp
-- Up-to-date: /usr/local/include//msp/msp_msg.hpp
-- Up-to-date: /usr/local/include//msp/ByteVector.hpp
-- Up-to-date: /usr/local/lib/pkgconfig/msp.pc

======================== Installation successful ==========================

Some of the files created by the installation are inside the home directory: /home

You probably don't want them to be included in the package.
Do you want me to list them?  [n]: n
Should I exclude them from the package? (Saying yes is a good idea)  [n]: y

Copying files to the temporary directory...OK

Stripping ELF binaries and libraries...OK

Compressing man pages...OK

Building file list...OK

Building Debian package...OK

Installing Debian package...OK

Erasing temporary files...OK

Writing backup package...OK
OK

Deleting temp dir...OK

**********************************************************************

 Done. The new package has been installed and saved to

 /home/kevin/Desktop/msp-3.0.0/build/build_20200303-1_amd64.deb

 You can remove it from your system anytime using: 

      dpkg -r build

**********************************************************************

kevin@kevin-MacBookPro:~/Desktop/msp-3.0.0/build$ 

And the output from making the example I sent you: 

make
g++ -O0 -g    -std=c++17 -c main.cpp -o main.o
g++ -O0 -g    -std=c++17 -c flight.cpp -o flight.o
g++ -O0 -g    -pthread main.o flight.o -o test -L/usr/local/lib/ -lmsp_fcu -lmspclient
flight.o: In function `std::ostream& operator<< <unsigned char>(std::ostream&, msp::Value<unsigned char> const&)':
/usr/local/include/msp/Value.hpp:77: multiple definition of `std::ostream& operator<< <unsigned char>(std::ostream&, msp::Value<unsigned char> const&)'
main.o:/usr/local/include/msp/Value.hpp:77: first defined here
flight.o: In function `std::ostream& operator<< <signed char>(std::ostream&, msp::Value<signed char> const&)':
/usr/local/include/msp/Value.hpp:86: multiple definition of `std::ostream& operator<< <signed char>(std::ostream&, msp::Value<signed char> const&)'
main.o:/usr/local/include/msp/Value.hpp:86: first defined here
flight.o: In function `std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::vector(std::initializer_list<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)':
/usr/local/include/msp/ByteVector.hpp:456: multiple definition of `operator<<(std::ostream&, msp::ByteVector const&)'
main.o:/usr/local/include/msp/ByteVector.hpp:456: first defined here
flight.o: In function `msp::firmwareVariantToString[abi:cxx11](msp::FirmwareVariant)':
/usr/local/include/msp/FirmwareVariants.hpp:36: multiple definition of `msp::firmwareVariantToString[abi:cxx11](msp::FirmwareVariant)'
main.o:/usr/local/include/msp/FirmwareVariants.hpp:36: first defined here
flight.o: In function `operator<<(std::ostream&, msp::ID const&)':
/usr/local/include/msp/Message.hpp:73: multiple definition of `operator<<(std::ostream&, msp::ID const&)'
main.o:/usr/local/include/msp/Message.hpp:73: first defined here
flight.o: In function `operator<<(std::ostream&, msp::Message const&)':
/usr/local/include/msp/Message.hpp:78: multiple definition of `operator<<(std::ostream&, msp::Message const&)'
main.o:/usr/local/include/msp/Message.hpp:78: first defined here
flight.o: In function `msp::armingFlagToString[abi:cxx11](unsigned int)':
/usr/local/include/msp/msp_msg.hpp:264: multiple definition of `msp::armingFlagToString[abi:cxx11](unsigned int)'
main.o:/usr/local/include/msp/msp_msg.hpp:264: first defined here
flight.o: In function `operator<<(std::ostream&, msp::msg::ImuSI const&)':
/usr/local/include/msp/msp_msg.hpp:5499: multiple definition of `operator<<(std::ostream&, msp::msg::ImuSI const&)'
main.o:/usr/local/include/msp/msp_msg.hpp:5499: first defined here
collect2: error: ld returned 1 exit status
Makefile:28: recipe for target 'test' failed
make: *** [test] Error 1

pollito5, 2020-03-04:
I got it to work. I was using your last release at first. I pulled the mast branch and reinstalled and it compiled perfectly. Thank you!

closed 2020-03-04 by pollito5

christianrauch, 2020-03-04:
Good to hear.
I see from your log that you used version 3.0.0. The latest version is 3.3.0 (see _Show 3 newer tags_ at https://github.com/christianrauch/msp/releases).

Better use the latest development version to check if an issue has been fixed. I also recently encountered a linking issue with newer gcc versions (#39).

Btw, I fixed the formatting of your terminal output. Please use code formatting (https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to format code and terminal output with monospace. It is much easier to read.