christianrauch/msp #42: Having trouble reaching desired read speeds

martindeegan, 2020-03-02:
Hi, thanks for the library. I'm trying to read IMU and motor commands on my Betaflight board from a Linux computer, but I'm having trouble getting over 100Hz. I have set `serial_update_rate_hz=2000`. Increasing the baudrate to 1000000 decreased throughput and 500000 was better.

Do you think this just a result of a poor cable? I will eventually want to write control commands to the board so speeding up the connection is really important for me.

christianrauch, 2020-03-03:
There are multiple factors that limit throughput. I certainly depends on the available bandwidth and the number of subscribed message types and their individual update rates.

I used to establish the serial connection via an integrated USB-to-serial converter directly on the flight controller board (e.g. see the Naze 32 Revision 6). This way, the serial connection is less affected by inference.

Apart from this, the update rate is limited by the compute capabilities on either end. I had issues achieving high update rates with a Raspberry Pi Zero because the process (in my case a ROS2 node) would max out the CPU. Also, I did not follow the development of betaflight recently. There might have been changes that require additional tuning.

At some point, I was able to achieve a 1kHz update rate for a simple message type (I don't remember which). This certainly would drop as soon as I subscribe to a second message type.

christianrauch, 2020-03-31:
@martindeegan Could you resolve your issue?

closed 2020-05-28 by christianrauch