christianrauch/msp #44: Fatal exception on website

MD-AZMAL, 2020-04-13:
http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol gives the follow error 
[98c64343] 2020-04-13 09:43:56: Fatal exception of type MWException

i am not sure if this issue belongs here but previously the link was running. so if you find this issue unsolicited please close it and guide me to right resources.

christianrauch, 2020-04-13:
I noticed that the wiki is not available a couple of weeks ago. I thought that is just temporary. If you need access to the wiki, you better get in contact with the maintainers of http://www.multiwii.com.

Alternatively, a snapshot of the wiki is available at: http://web.archive.org/web/20190812122529/http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol.

MD-AZMAL, 2020-04-13:
Thanks for help

closed 2020-04-13 by MD-AZMAL