christianrauch/msp #45: Build works fine but compilation gives error

MD-AZMAL, 2020-04-13:
On compiling fcu_test using 
```
gcc fcu_test.cpp
```

i get the following error

```
In file included from /usr/include/asio/associated_allocator.hpp:18,
                 from /usr/include/asio.hpp:18,
                 from /usr/include/c++/9/Client.hpp:4,
                 from /usr/include/c++/9/FlightController.hpp:4,
                 from fcu_test.cpp:1:
/usr/include/asio/detail/config.hpp:26:11: fatal error: boost/config.hpp: No such file or directory
   26 | # include <boost/config.hpp>
      |           ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

I have installed all the necessary msp related header files but still this asio issue pops up. I checked your comment https://github.com/christianrauch/msp/issues/4#issuecomment-301948254 which says it does not depend on boost then why am i getting this error?

christianrauch, 2020-04-13:
How does the "build" work but the "compilation" fails? Usually, the built includes the compilation.

You directly try to compile the C source file without providing all the necessary build and linker flags. It won't work like this. You have to follow the [build instructions](https://github.com/christianrauch/msp#installation-and-test) and use CMake to build the project.

If you really want to build this without CMake, you have to provide all the flags, such as [`ASIO_STANDALONE`](https://think-async.com/Asio/AsioStandalone.html), manually.

MD-AZMAL, 2020-04-13:
okay i will try that, keeping the issue still open just in case it requires further querying.

MD-AZMAL, 2020-04-13:
Apparently ```ASIO_STANDALONE``` parameter did not work ```DASIO_STADALONE``` was the one working. But when compiling with c++11 standard i get this error.
```
g++ -std=c++11 fcu_test.cpp -DASIO_STANDALONE
```
Error : 
```
In file included from /usr/include/c++/9/FlightController.hpp:4,
                 from fcu_test.cpp:1:
/usr/include/c++/9/Client.hpp: In member function ‘std::shared_ptr<msp::client::SubscriptionBase> msp::client::Client::subscribe(const std::function<void(const T&)>&, const double&)’:
/usr/include/c++/9/Client.hpp:163:48: error: ‘make_unique’ is not a member of ‘std’
  163 |             recv_callback, send_callback, std::make_unique<T>(fw_variant), tp);
      |                                                ^~~~~~~~~~~
/usr/include/c++/9/Client.hpp:163:48: note: ‘std::make_unique’ is only available from C++14 onwards
/usr/include/c++/9/Client.hpp:163:61: error: expected primary-expression before ‘>’ token
  163 |             recv_callback, send_callback, std::make_unique<T>(fw_variant), tp);
      |                                                             ^
In file included from /usr/include/c++/9/FlightController.hpp:7,
                 from fcu_test.cpp:1:
/usr/include/c++/9/msp_msg.hpp: In member function ‘virtual msp::ByteVectorUptr msp::msg::SetInavPid::encode() const’:
/usr/include/c++/9/msp_msg.hpp:577:36: error: ‘make_unique’ is not a member of ‘std’
  577 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                    ^~~~~~~~~~~
/usr/include/c++/9/msp_msg.hpp:577:36: note: ‘std::make_unique’ is only available from C++14 onwards
/usr/include/c++/9/msp_msg.hpp:577:58: error: expected primary-expression before ‘>’ token
  577 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                          ^
/usr/include/c++/9/msp_msg.hpp:577:60: error: expected primary-expression before ‘)’ token
  577 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                            ^
/usr/include/c++/9/msp_msg.hpp: In member function ‘virtual msp::ByteVectorUptr msp::msg::SetBoardName::encode() const’:
/usr/include/c++/9/msp_msg.hpp:616:36: error: ‘make_unique’ is not a member of ‘std’
  616 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                    ^~~~~~~~~~~
/usr/include/c++/9/msp_msg.hpp:616:36: note: ‘std::make_unique’ is only available from C++14 onwards
/usr/include/c++/9/msp_msg.hpp:616:58: error: expected primary-expression before ‘>’ token
  616 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                          ^
/usr/include/c++/9/msp_msg.hpp:616:60: error: expected primary-expression before ‘)’ token
  616 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                            ^
/usr/include/c++/9/msp_msg.hpp: In member function ‘virtual msp::ByteVectorUptr msp::msg::SetNavPosHold::encode() const’:
/usr/include/c++/9/msp_msg.hpp:660:36: error: ‘make_unique’ is not a member of ‘std’
  660 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                    ^~~~~~~~~~~
/usr/include/c++/9/msp_msg.hpp:660:36: note: ‘std::make_unique’ is only available from C++14 onwards
/usr/include/c++/9/msp_msg.hpp:660:58: error: expected primary-expression before ‘>’ token
  660 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                          ^
/usr/include/c++/9/msp_msg.hpp:660:60: error: expected primary-expression before ‘)’ token
  660 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                            ^
/usr/include/c++/9/msp_msg.hpp: In member function ‘virtual msp::ByteVectorUptr msp::msg::SetCalibrationData::encode() const’:
/usr/include/c++/9/msp_msg.hpp:712:36: error: ‘make_unique’ is not a member of ‘std’
  712 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                    ^~~~~~~~~~~
/usr/include/c++/9/msp_msg.hpp:712:36: note: ‘std::make_unique’ is only available from C++14 onwards
/usr/include/c++/9/msp_msg.hpp:712:58: error: expected primary-expression before ‘>’ token
  712 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                          ^
/usr/include/c++/9/msp_msg.hpp:712:60: error: expected primary-expression before ‘)’ token
  712 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                            ^
/usr/include/c++/9/msp_msg.hpp: In member function ‘virtual msp::ByteVectorUptr msp::msg::SetPositionEstimationConfig::encode() const’:
/usr/include/c++/9/msp_msg.hpp:767:36: error: ‘make_unique’ is not a member of ‘std’
  767 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                    ^~~~~~~~~~~
/usr/include/c++/9/msp_msg.hpp:767:36: note: ‘std::make_unique’ is only available from C++14 onwards
/usr/include/c++/9/msp_msg.hpp:767:58: error: expected primary-expression before ‘>’ token
  767 |         ByteVectorUptr data = std::make_unique<ByteVector>();
      |                                                          ^
[... 724 lines cut ...]
[... cut ...]
In file included from /usr/include/asio/associated_allocator.hpp:20,
                 from /usr/include/asio.hpp:18,
                 from /usr/include/c++/9/Client.hpp:4,
                 from /usr/include/c++/9/FlightController.hpp:4,
                 from fcu_test.cpp:1:
/usr/include/asio/detail/type_traits.hpp:23:11: fatal error: boost/type_traits/add_const.hpp: No such file or directory
   23 | # include <boost/type_traits/add_const.hpp>
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I am not sure what to do.

christianrauch, 2020-04-13:
> I am not sure what to do.

Follow the build instructions and use CMake :-)

christianrauch, 2020-04-14:
@MD-AZMAL Did you manage to compile the project using CMake according to the instructions? If you still have problems with the CMake build instructions, let me know.

MD-AZMAL, 2020-04-15:
No CMake works fine actually i was trying to use nodejs native addon so i can compile endpoints for my javascript project. Thus i need a way to compile my code using g++. Once that is done i would then move to build it using node gyp so that these methods are accessible in javascript.

MD-AZMAL, 2020-04-15:
unfortunately i did not find any good javascript libraries that can be used for this purpose. Most of them are outdated and some do not function well. If you know anything that would help me please mention.

christianrauch, 2020-04-15:
You want to generate JavaScript bindings for the msp library?

Is there a reason you have to use plain g++ for this, instead of using some build tool like CMake? I think it would be much easier to extend the CMake project with the wrapper, than trying to replicate the CMake script behaviour. I am happy to merge that functionality into this repo if you get it working.

Would [cmake-js](https://github.com/cmake-js/cmake-js) help?

MD-AZMAL, 2020-04-16:
yes i was trying to make javascript bindings! true maybe i should try use cmake instead. Let's see where i get :P, i'll have a look at https://github.com/cmake-js/cmake-js thanks for pointing it out!

christianrauch, 2020-04-16:
Can I then close this issue for now? The binding integration is not exactly a build issue with the CMake project.

If you want to add the JavaScript bindings to the CMake project, you can open a PR with this functionality and we can fix any outstanding issues there.

MD-AZMAL, 2020-04-17:
yes ofcourse.. feel free to close the issue! 😁

closed 2020-04-17 by christianrauch