christianrauch/msp #50: I have not tested the `MSP_SET_WP` message before. If you provide a minimal source example to reproduce this issue (as plain text, not screenshots of text) I can try to reproduce this in my Betaflight setup.

SALTU-19, 2020-09-02:
I have not tested the `MSP_SET_WP` message before. If you provide a minimal source example to reproduce this issue (as plain text, not screenshots of text) I can try to reproduce this in my Betaflight setup.

You can increase the logger level to debug to get more informative output. Looking at the source code, receiving this error indicates that `ok_id` is false, which means that the direction byte is `!`. This is typically the case if the FCU does not support the requested MSP ID.

Your EOF message may further indicate a connection error. A request with an unsupported MSP ID should not cause connection issues. Have you encountered other connection issues over your serial connection?

Can you:
1. Test this with an MSPv1 connection instead of the MSPv2 connection.
2. Verify that this message works correctly with the iNav GUI with MSP version 1 and 2 connections.
3. Test the connection via a USB connection (USB-to-Serial adapter on FCU) to rule out disturbances with unshielded plain serial connections.

_Originally posted by @christianrauch in https://github.com/christianrauch/msp/issues/49#issuecomment-685206486_

SALTU-19, 2020-09-02:
`#include <FlightController.hpp>
#include <iostream>
#include <msp_msg.hpp>

#define MSP_NAV_STATUS_WAYPOINT_ACTION_WAYPOINT 0x01

struct SetWp : public Message {
    SetWp(FirmwareVariant v) : Message(v) {}

    virtual ID id() const override { return ID::MSP_SET_WP; }

    Value<uint8_t> wp_no;
    Value<uint8_t> action;
    Value<uint32_t> lat;
    Value<uint32_t> lon;
    Value<uint32_t> alt;

    Value<uint16_t> p1;
    Value<uint16_t> p2;
    Value<uint16_t> p3;
    Value<uint8_t> nav_flag;

    virtual ByteVectorUptr encode() const override {
        ByteVectorUptr data = std::make_unique<ByteVector>();
        bool rc             = true;
        rc &= data->pack(wp_no);
	rc &= data->pack(action);
        rc &= data->pack(lat);
        rc &= data->pack(lon);
        rc &= data->pack(alt);
        rc &= data->pack(p1);
        rc &= data->pack(p2);
        if(fw_variant == FirmwareVariant::INAV) {
            rc &= data->pack(p3);
        }
        rc &= data->pack(nav_flag);
        return data;
    }
};

int main(int argc, char* argv[]) {
    msp::FirmwareVariant fw_variant = msp::FirmwareVariant::INAV;
    const std::string device =
        (argc > 1) ? std::string(argv[1]) : "/dev/ttySUB0";
    const size_t baudrate = (argc > 2) ? std::stoul(argv[2]) : 115200;

    fcu::FlightController fcu;
    fcu.setLoggingLevel(msp::client::LoggingLevel::INFO);
    // wait for connection
    fcu.connect(device, baudrate);

    msp::msg::SetWp setwp(fw_variant);

    
	setwp.wp_no = 255;
	setwp.action = MSP_NAV_STATUS_WAYPOINT_ACTION_WAYPOINT;
	setwp.lat = 40811854;
	setwp.lon = 29360025;
	setwp.alt = 1500;
	setwp.p1 = 80;
	setwp.p2 = 0;
	setwp.p3 = 0;
	setwp.nav_flag = 0;

	if (fcu.sendMessage(setwp) ) {
		std::cout << "Acknowled";
	}
	else
		std::cout << "wasted";

}

`
This is the final version of my code.

how can i test MSPv1 connection ?

christianrauch, 2020-09-02:
Is this separate issue intended? You have the same information in https://github.com/christianrauch/msp/issues/49#issuecomment-685704184.

SALTU-19, 2020-09-02:
I guess I opened it by mistake. :)

closed 2020-09-02 by SALTU-19