christianrauch/msp #59: cannot connect to betaflight fcs

catkira, 2024-02-26:
I get this error
```
#FC variant:
 Identifier: BTFL
#Api Version:
 API: 1.46
 Protocol: 0
#FC version:
 Version: 4.5.0
#Board Info:
 Identifier: S7X2
 Version: 0
 OSD support: 2
 Comms bitmask: 67
 Board Name: STM32F7X2
#Build Info:
 Date: Jan 13 2024
 Time: 14:39:30
 Git revision: f1cbd83
#Status:
 Cycle time: 124 us
 I2C errors: 0
 Sensors:
    Accelerometer: ON
    Barometer: OFF
    Magnetometer: OFF
    GPS: OFF
    Sonar: OFF
 Active Boxes (by ID):
#Ident:
 MultiWii Version: <unset>
 MSP Version: <unset>
 Type: UNDEFINED
 Capabilities:
    Bind:   OFF
    DynBal: OFF
    Flap:   OFF
Message marker 65 is not recognised!
Version marker 85 is not recognised!
terminate called after throwing an instance of 'std::runtime_error'
  what():  Cannot get BoxNames!
```

catkira, 2024-02-26:
with log level DEBUG I get this output
```
sending message - ID 44
sending: 44 |
packed: 24 4d 3c 0 2c 2c
write complete: 0 vs 6
message failed to send
client_.sendMessage(rxConfig) failed
#RX configuration:
 serialrx_provider: <unset>
 maxcheck: <unset>
 midrc: <unset>
 mincheck: <unset>
 spektrum_sat_bind: <unset>
 rx_min_usec: <unset>
 rx_max_usec: <unset>
 rcInterpolation: <unset>
 rcInterpolationInterval: <unset>
 airModeActivateThreshold: <unset>
 rx_spi_protocol: <unset>
 rx_spi_id: <unset>
 rx_spi_rf_channel_count: <unset>
 fpvCamAngleDegrees: <unset>
 receiverType: <unset>
sending message - ID 45
sending: 45 |
packed: 24 4d 3c 0 2d 2d
write complete: 0 vs 6
message failed to send
sending message - ID 2
sending: 2 |
packed: 24 4d 3c 0 2 2
write complete: 6 vs 6
processOneMessage on 10 bytes
processOneMessage finished
#FC variant:
 Identifier: BTFL
sending message - ID 1
sending: 1 |
packed: 24 4d 3c 0 1 1
write complete: 6 vs 6
processOneMessage on 9 bytes
processOneMessage finished
#Api Version:
 API: 1.46
 Protocol: 0
sending message - ID 3
sending: 3 |
packed: 24 4d 3c 0 3 3
write complete: 6 vs 6
processOneMessage on 9 bytes
processOneMessage finished
#FC version:
 Version: 4.5.0
sending message - ID 4
sending: 4 |
packed: 24 4d 3c 0 4 4
write complete: 6 vs 6
processOneMessage on 83 bytes
processOneMessage finished
#Board Info:
 Identifier: S7X2
 Version: 0
 OSD support: 2
 Comms bitmask: 67
 Board Name: STM32F7X2
sending message - ID 5
sending: 5 |
packed: 24 4d 3c 0 5 5
write complete: 6 vs 6
processOneMessage on 32 bytes
processOneMessage finished
#Build Info:
 Date: Jan 13 2024
 Time: 14:39:30
 Git revision: f1cbd83
sending message - ID 101
sending: 101 |
packed: 24 4d 3c 0 65 65
write complete: 6 vs 6
processOneMessage on 30 bytes
processOneMessage finished
#Status:
 Cycle time: 124 us
 I2C errors: 0
 Sensors:
    Accelerometer: ON
    Barometer: OFF
    Magnetometer: OFF
    GPS: OFF
    Sonar: OFF
 Active Boxes (by ID):
sending message - ID 100
sending: 100 |
packed: 24 4d 3c 0 64 64
write complete: 6 vs 6
processOneMessage on 6 bytes
Message v1 with ID 100 is not recognised!
processOneMessage finished
#Ident:
 MultiWii Version: <unset>
 MSP Version: <unset>
 Type: UNDEFINED
 Capabilities:
    Bind:   OFF
    DynBal: OFF
    Flap:   OFF
sending message - ID 116
sending: 116 |
packed: 24 4d 3c 0 74 74
write complete: 6 vs 6
processOneMessage on 261 bytes
Message v1 with ID 116 has wrong CRC! (expected: 149, received: 76)
processOneMessage finished
processOneMessage on 3 bytes
Message marker 65 is not recognised!
Version marker 85 is not recognised!
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
terminate called after throwing an instance of 'std::runtime_error'
  what():  Cannot get BoxNames!
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
buffer returned EOF; reading char directly from port
Aborted
```

catkira, 2024-02-26:
There could be a problem if payload length is more than 255 bytes
[image]

catkira, 2024-02-26:
In betaflight source code I can see that message size can be larger than 255 bytes
[image]

defrag-bambino, 2024-02-26:
Hello,

I am having the same issue right now. For me, it works fine if I NOT set msp_override_channels_mask = 245760, but keep it at = 0.

catkira, 2024-02-26:
I think you probably had another issue. I could fix my issue with this  PR: https://github.com/christianrauch/msp/pull/60

The problem was that betaflight sent a message with payload size > 255 which msp could not handle.

defrag-bambino, 2024-02-26:
Sadly it does not seem to fix it? Using your PR it processes the "jumbo" message but is stuck afterwards and does not continue.
```
./build/msp/fcu_test /dev/ttyACM0
sending message - ID 2
sending: 2 | 
packed: 24 4d 3c 0 2 2 
write complete: 6 vs 6
processOneMessage on 10 bytes
processOneMessage finished
sending message - ID 1
sending: 1 | 
packed: 24 4d 3c 0 1 1 
write complete: 6 vs 6
processOneMessage on 9 bytes
processOneMessage finished
sending message - ID 4
sending: 4 | 
packed: 24 4d 3c 0 4 4 
write complete: 6 vs 6
processOneMessage on 87 bytes
processOneMessage finished
sending message - ID 101
sending: 101 | 
packed: 24 4d 3c 0 65 65 
write complete: 6 vs 6
processOneMessage on 28 bytes
processOneMessage finished
sending message - ID 100
sending: 100 | 
packed: 24 4d 3c 0 64 64 
write complete: 6 vs 6
processOneMessage on 6 bytes
Message v1 with ID 100 is not recognised!
processOneMessage finished
sending message - ID 116
sending: 116 | 
packed: 24 4d 3c 0 74 74 
write complete: 6 vs 6
received jumbo message with payload_size 65535 available bytes 263
```

catkira, 2024-02-26:
hmmm, `payload_size 65535` looks wrong, you can try do debug :)

catkira, 2024-02-26:
I get this
```
sending message - ID 116
sending: 116 |
packed: 24 4d 3c 0 74 74
write complete: 6 vs 6
processOneMessage on 317 bytes
received jumbo message with length 309
sending message - ID 119
sending: 119 |
packed: 24 4d 3c 0 77 77
write complete: 6 vs 6
processOneMessage finished
processOneMessage on 32 bytes
processOneMessage finished
sending message - ID 64
sending: 64 |
packed: 24 4d 3c 0 40 40
write complete: 6 vs 6
```

catkira, 2024-02-26:
I did some force pushes. Can you post ur git hash?

catkira, 2024-02-26:
that's what I am using too. You have to debug where this wrong 65535 size comes from. It should be 317.

catkira, 2024-02-26:
but it could be different for your betaflight version and settings. I am using betaflight 4.5 from github.

defrag-bambino, 2024-02-26:
I am using official Betaflight 4.4.3 on a SpeedyBee F405 V3

catkira, 2024-02-26:
it would be nice if you can debug the problem and improve my PR

defrag-bambino, 2024-02-26:
do you have "msp_override_channels_mask" set to anything other than 0?

catkira, 2024-02-26:
I did not change that value

defrag-bambino, 2024-02-26:
try changing it. Mine is 245760. Thats what breaks it for me

catkira, 2024-02-26:
maybe its a betaflight bug and it outputs garbage in that case ^^

catkira, 2024-02-26:
hmm looks like a betaflight bug? Can u add a debug output of the raw rx bytes and open an issue in betaflight github?

defrag-bambino, 2024-02-26:
I tried it using YAMSPY and that works fine :thinking:

defrag-bambino, 2024-02-26:
By the way, the Betaflight FC sends the FlightControllerIdentifier "BTFL", but in [FirmwareVariants.hpp](https://github.com/christianrauch/msp/blob/master/inc/msp/FirmwareVariants.hpp) Betaflight is called "BAFL". Do you think this is a problem?

catkira, 2024-02-26:
I saw that too, I think it's a bug, but I did not see that it causes any problems besides displaying BAFL instead of BTFL.

defrag-bambino, 2024-02-26:
Where can I print the raw RX bytes? :see_no_evil:

catkira, 2024-02-26:
U have to add debug prints in source code. See how the other debug messages are printed.

defrag-bambino, 2024-02-26:
Yeah but where do the raw RX bytes come in? What line?

defrag-bambino, 2024-02-26:
ok. does yours also not work with the msp_override_channels_mask set?

catkira, 2024-02-26:
I will try it when I need it :)

defrag-bambino, 2024-02-26:
is this the bytes?
```

#Attitude:
 Roll : 0 deg
 Pitch : 0 deg
 Heading: 353 deg
sending message - ID 109
sending: 109 | 
packed: 24 4d 3c 0 6d 6d 
write complete: 6 vs 6
processOneMessage on 12 bytes
processOneMessage finished
---------------e2 ff ff ff 0 0 

#Altitude:
 Altitude: -0.3 m, var: 0 m/s
 Barometer: <unset>
sending message - ID 110
sending: 110 | 
packed: 24 4d 3c 0 6e 6e 
write complete: 6 vs 6
processOneMessage on 15 bytes
processOneMessage finished
---------------0 0 0 0 0 0 0 1 0 

#Analog:
 Battery Voltage: 0 V
 Current: 0 A
 Power consumption: 0 Ah
 RSSI: 0
sending message - ID 111
sending: 111 | 
packed: 24 4d 3c 0 6f 6f 
write complete: 6 vs 6
processOneMessage on 29 bytes
processOneMessage finished
---------------7 0 43 43 43 0 32 0 0 0 0 7 7 0 0 64 ce 7 ce 7 ce 7 3 

Print method for message ID 111 is not implemented
sending message - ID 112
sending: 112 | 
packed: 24 4d 3c 0 70 70 
write complete: 6 vs 6
processOneMessage on 21 bytes
processOneMessage finished
---------------2d 50 28 2f 54 2e 2d 50 0 32 32 4b 28 0 0 

unsupported: 112
sending message - ID 113
sending: 113 | 
packed: 24 4d 3c 0 71 71 
write complete: 6 vs 6
processOneMessage on 6 bytes
Message v1 with ID 113 is not recognised!
processOneMessage finished
unsupported: 113
sending message - ID 114
sending: 114 | 
packed: 24 4d 3c 0 72 72 
write complete: 6 vs 6
processOneMessage on 6 bytes
Message v1 with ID 114 is not recognised!
processOneMessage finished
unsupported: 114
sending message - ID 115
sending: 115 | 
packed: 24 4d 3c 0 73 73 
write complete: 6 vs 6
processOneMessage on 6 bytes
Message v1 with ID 115 is not recognised!
processOneMessage finished
unsupported: 115
sending message - ID 116
sending: 116 | 
packed: 24 4d 3c 0 74 74 
write complete: 6 vs 6
received jumbo message with payload_size 65535 available bytes 128
received jumbo message with payload_size 65535 available bytes 263

```

catkira, 2024-02-26:
yes, but the byte output from the problematic 116 message is not there. I have to work on other stuff now...

defrag-bambino, 2024-02-26:
ok I will try to find the issue, thanks!

catkira, 2024-02-26:
There were recent BF bugfixes related to MSP, such as https://github.com/betaflight/betaflight/pull/12857

It would be good if you can test with BF 4.5 RC2 if the problem is still there.

defrag-bambino, 2024-02-26:
Ok. I will try. I just printed the bytes:
```
.......
processOneMessage finished
unsupported: 114
sending message - ID 115
sending: 115 | 
packed: 24 4d 3c 0 73 73 
write complete: 6 vs 6
Message type: $M
Payload byte: 4d
Payload byte: 21
Payload size: 0
Payload byte: 73
Payload byte: 73
processOneMessage on 6 bytes
Message v1 with ID 115 is not recognised!
processOneMessage finished
unsupported: 115
sending message - ID 116
sending: 116 | 
packed: 24 4d 3c 0 74 74 
write complete: 6 vs 6
Message type: $M
Payload byte: 4d
Payload byte: 3e
Payload size: 255
Payload byte: 74
Payload byte: ff
Payload byte: 00
Payload byte: 41
Payload byte: 52
Payload byte: 4d
Payload byte: 3b
Payload byte: 41
Payload byte: 4e
Payload byte: 47
Payload byte: 4c
Payload byte: 45
Payload byte: 3b
Payload byte: 48
Payload byte: 4f
Payload byte: 52
Payload byte: 49
Payload byte: 5a
Payload byte: 4f
Payload byte: 4e
Payload byte: 3b
Payload byte: 48
Payload byte: 45
Payload byte: 41
Payload byte: 44
Payload byte: 46
Payload byte: 52
Payload byte: 45
Payload byte: 45
Payload byte: 3b
Payload byte: 46
Payload byte: 41
Payload byte: 49
Payload byte: 4c
Payload byte: 53
Payload byte: 41
Payload byte: 46
Payload byte: 45
Payload byte: 3b
Payload byte: 48
Payload byte: 45
Payload byte: 41
Payload byte: 44
Payload byte: 41
Payload byte: 44
Payload byte: 4a
Payload byte: 3b
Payload byte: 42
Payload byte: 45
Payload byte: 45
Payload byte: 50
Payload byte: 45
Payload byte: 52
Payload byte: 3b
Payload byte: 4f
Payload byte: 53
Payload byte: 44
Payload byte: 20
Payload byte: 44
Payload byte: 49
Payload byte: 53
Payload byte: 41
Payload byte: 42
Payload byte: 4c
Payload byte: 45
Payload byte: 3b
Payload byte: 42
Payload byte: 4c
Payload byte: 41
Payload byte: 43
Payload byte: 4b
Payload byte: 42
Payload byte: 4f
Payload byte: 58
Payload byte: 3b
Payload byte: 46
Payload byte: 50
Payload byte: 56
Payload byte: 20
Payload byte: 41
Payload byte: 4e
Payload byte: 47
Payload byte: 4c
Payload byte: 45
Payload byte: 20
Payload byte: 4d
Payload byte: 49
Payload byte: 58
Payload byte: 3b
Payload byte: 43
Payload byte: 41
Payload byte: 4d
Payload byte: 45
Payload byte: 52
Payload byte: 41
Payload byte: 20
Payload byte: 43
Payload byte: 4f
Payload byte: 4e
Payload byte: 54
Payload byte: 52
Payload byte: 4f
Payload byte: 4c
Payload byte: 20
Payload byte: 31
Payload byte: 3b
Payload byte: 43
Payload byte: 41
Payload byte: 4d
Payload byte: 45
Payload byte: 52
Payload byte: 41
Payload byte: 20
Payload byte: 43
Payload byte: 4f
Payload byte: 4e
Payload byte: 54
Payload byte: 52
Payload byte: 4f
Payload byte: 4c
Payload byte: 20
Payload byte: 32
Payload byte: 3b
Payload byte: 43
Payload byte: 41
Payload byte: 4d
Payload byte: 45
Payload byte: 52
Payload byte: 41
Payload byte: 20
Payload byte: 43
Payload byte: 4f
Payload byte: 4e
Payload byte: 54
Payload byte: 52
Payload byte: 4f
Payload byte: 4c
Payload byte: 20
Payload byte: 33
Payload byte: 3b
Payload byte: 50
Payload byte: 52
Payload byte: 45
Payload byte: 41
Payload byte: 52
Payload byte: 4d
Payload byte: 3b
Payload byte: 56
Payload byte: 54
Payload byte: 58
Payload byte: 20
Payload byte: 50
Payload byte: 49
Payload byte: 54
Payload byte: 20
Payload byte: 4d
Payload byte: 4f
Payload byte: 44
Payload byte: 45
Payload byte: 3b
Payload byte: 50
Payload byte: 41
Payload byte: 52
Payload byte: 41
Payload byte: 4c
Payload byte: 59
Payload byte: 5a
Payload byte: 45
Payload byte: 3b
Payload byte: 56
Payload byte: 54
Payload byte: 58
Payload byte: 20
Payload byte: 43
Payload byte: 4f
Payload byte: 4e
Payload byte: 54
Payload byte: 52
Payload byte: 4f
Payload byte: 4c
Payload byte: 20
Payload byte: 44
Payload byte: 49
Payload byte: 53
Payload byte: 41
Payload byte: 42
Payload byte: 4c
Payload byte: 45
Payload byte: 3b
Payload byte: 4c
Payload byte: 41
Payload byte: 55
Payload byte: 4e
Payload byte: 43
Payload byte: 48
Payload byte: 20
Payload byte: 43
Payload byte: 4f
Payload byte: 4e
Payload byte: 54
Payload byte: 52
Payload byte: 4f
Payload byte: 4c
Payload byte: 3b
Payload byte: 4d
Payload byte: 53
Payload byte: 50
Payload byte: 20
Payload byte: 4f
Payload byte: 56
Payload byte: 45
Payload byte: 52
Payload byte: 52
Payload byte: 49
Payload byte: 44
Payload byte: 45
Payload byte: 3b
Payload byte: 53
Payload byte: 54
Payload byte: 49
Payload byte: 43
Payload byte: 4b
Payload byte: 20
Payload byte: 43
Payload byte: 4f
Payload byte: 4d
Payload byte: 4d
Payload byte: 41
Payload byte: 4e
Payload byte: 44
Payload byte: 53
Payload byte: 20
Payload byte: 44
Payload byte: 49
Payload byte: 53
Payload byte: 41
Payload byte: 42
Payload byte: 4c
Payload byte: 45
Payload byte: 3b
Payload byte: 42
Payload byte: 45
Payload byte: 45
Payload byte: 50
Payload byte: 45
Payload byte: 52
Payload byte: 20
Payload byte: 4d
Payload byte: 55
Payload byte: 54
Payload byte: 45
Payload byte: 3b
Payload byte: 52
Payload byte: 45
Payload byte: 41
Payload byte: 44
Payload byte: 59
Payload byte: 3b
Payload byte: 41
received jumbo message with payload_size 65535 available bytes 263
^C
```

catkira, 2024-02-26:
```
Payload byte: 00
Payload byte: 41
```
should be the payload length, 0 + 41 << 8 = 16640, that does not make sense. I also wonder why the debug message says `payload_size 65535` ...
I think you need to do a bit more debugging.

defrag-bambino, 2024-02-26:
Indeed, Betaflight 4.5 RC2 with your PR fixes the issue!
Thank you very much!

catkira, 2024-02-26:
ok, then it was a BF bug after all.

defrag-bambino, 2024-02-26:
yes, seems like it

closed 2024-02-26 by catkira

DangerD1024, 2024-09-23:
Betaflight 4.5 - Same issue here, how it can be fixed?

Coder-X15, 2025-03-01:
@catkira please refer #68 . I am facing a similar issue, it seems, plus it's my first time dealing with FCs and their firmwares.

Coder-X15, 2025-03-01:
> with log level DEBUG I get this output
>
> ```
> sending message - ID 44
> sending: 44 |
> packed: 24 4d 3c 0 2c 2c
> write complete: 0 vs 6
> message failed to send
> client_.sendMessage(rxConfig) failed
> #RX [...]

How do I turn on log level DEBUG?

christianrauch, 2025-03-01:
> How do I turn on log level DEBUG?

`Client::setLoggingLevel`

Coder-X15, 2025-03-01:
> `Client::setLoggingLevel`

Done that just now, I have posted the results in the thread I started.