christianrauch/msp #64: CLI Parameters (get/set)

DangerD1024, 2024-08-27:
After staring i need first to check parameters if they are correct:
`get serial_update_rate_hz
`if it's not 2000, set it:
```
set serial_update_rate_hz=2000
save
```
Can i achieve that somehow?

GeorgXIII, 2024-09-10:
Hi there!

Take a look at these [messages](https://github.com/christianrauch/msp/blob/3c575d6a1aac7c2253823fdf552af97168aebcc9/inc/msp/msp_msg.hpp#L207):
```
    MSP2_COMMON_SETTING          = 0x1003,  // in/out message, returns setting
    MSP2_COMMON_SET_SETTING      = 0x1004,  // in message, sets a setting value
```

DangerD1024, 2024-09-13:
> Hi there!
>
> Take a look at these [messages](https://github.com/christianrauch/msp/blob/3c575d6a1aac7c2253823fdf552af97168aebcc9/inc/msp/msp_msg.hpp#L207):
>
> ```
>     MSP2_COMMON_SETTING          = [...]

Can you point me on how can I use that?

GeorgXIII, 2024-09-16:
I implemented it this way (for INAV 6.1 iirc):
```
using FlightControllerSetting = std::variant<std::string, uint8_t, int8_t, uint16_t, int16_t, uint32_t, float>;

FlightControllerSetting FlightController::getSetting(const std::string& name, msp::msg::DATA_TYPE type) const
{
    msp::msg::CommonSetting setting(m_firmwareVariant);
    setting.setting_name = name;
    setting.expected_data_type = type;
    if (!m_client->sendMessage(setting)) {
        // TODO: error handling
    }
    switch (setting.expected_data_type) {
    case msp::msg::DATA_TYPE::STRING:
        return setting.string_val;
        break;
    case msp::msg::DATA_TYPE::UINT8:
        return setting.uint8_val;
        break;
    case msp::msg::DATA_TYPE::INT8:
        return setting.int8_val;
        break;
    case msp::msg::DATA_TYPE::UINT16:
        return setting.uint16_val;
        break;
    case msp::msg::DATA_TYPE::INT16:
        return setting.int16_val;
        break;
    case msp::msg::DATA_TYPE::UINT32:
        return setting.uint32_val;
        break;
    case msp::msg::DATA_TYPE::FLOAT:
        return setting.float_val;
        break;
    }
    // TODO: error handling here
}

void FlightController::setSetting(const std::string& name, const FlightControllerSetting& value) const
{
    msp::msg::CommonSetSetting setting(m_firmwareVariant);
    setting.setting_name = name;
    if (const std::string* pval = std::get_if<std::string>(&value)) {
        setting.string_val = *pval;
    } else if (const uint8_t* pval = std::get_if<uint8_t>(&value)) {
        setting.uint8_val = *pval;
    } else if (const int8_t* pval = std::get_if<int8_t>(&value)) {
        setting.int8_val = *pval;
    } else if (const uint16_t* pval = std::get_if<uint16_t>(&value)) {
        setting.uint16_val = *pval;
    } else if (const int16_t* pval = std::get_if<int16_t>(&value)) {
        setting.int16_val = *pval;
    } else if (const uint32_t* pval = std::get_if<uint32_t>(&value)) {
        setting.uint8_val = *pval;
    } else if (const float* pval = std::get_if<float>(&value)) {
        setting.uint8_val = *pval;
    }

    if (!m_client->sendMessage(setting)) {
        // TODO: error handling here
    }
}
```

DangerD1024, 2024-09-20:
setSetting("roll_srate",67);
How can i set "aux 1 1 1 1300 2100 0 0"
And how can i call "save" to save parameters?

GeorgXIII, 2024-09-20:
If you want to set Mode range and channel, I think you should use MSP_SET_MODE_RANGE message

DangerD1024, 2024-09-20:
Betaflight 4.5 - Cannot get BoxNames! error on connect... investigating...