christianrauch/msp #68: Interfacing a Kakute F722 with Jetson Nano

Coder-X15, 2025-02-27:
I am new to using a Jetson Nano to control a flight controller, and as a result I was hunting through the internet when I found this. I did everything except setting up the serial update task rate (which I let to remain in the default 115200Hz). When I try to run every test code in the repo, I am getting CRC errors. For instance:

[image]

My few doubts are:
1. Why are the commands not being recognised?
2. I am using a 5V 3A power supply (basically my 44W Vivo mobile charger, I have nothing else to substitute it with) through the USB-C port on the Nano, so is the power supply an issue, possibly like slowing down the serial task due to power consumption constraints?
3. Can you suggest a fix for this?
4. Also, how I can I use this inside ROS1?

Coder-X15, 2025-02-27:
FYI I used different rates later on, it wasn't even working with the 115200Hz as well as the others. It must have been insensible of me to use the smaller ones due to the rate mismatch but IDK why it isn't working with the original.

christianrauch, 2025-02-27:
The examples take the baudrate as the second argument. You say your FC is using 115200, but for the examples you use different rates, e.g. "25600" and "100". This won't work.

Coder-X15, 2025-03-01:
> The examples take the baudrate as the second argument. You say your FC is using 115200, but for the examples you use different rates, e.g. "25600" and "100". This won't work.
> 
> For the other error, this looks like you are connected, but the FC does not know the message types `100` and `116`. Maybe you are using a newer FC version where these messages were removed. I could only test with older versions. But you should be able to subscribe to the other messages types. Feel free to adjust the example and remove those messages that are not supported.
> 
> Regarding ROS integration: This is just a C++ library, you can use it any way you like. The examples give you a good overview of how to instantiate and use a `fcu::FlightController` to read data and send messages. Other than this, I would strongly advise against using the older ROS 1 since this will soon lose support and become deprecated. If you really must use ROS 1, have a look at https://github.com/christianrauch/ros-multiwii, and https://github.com/christianrauch/multiwii_ros2 for ROS 2. These nodes are rather trivial since they just forward the messages to/from the library with some conversion. So you can also just write your own node using them as a base.

I did test with 115200 tbh, my screenshot didn't have it. I apologise for the confusion.

Coder-X15, 2025-03-01:
> The examples take the baudrate as the second argument. You say your FC is using 115200, but for the examples you use different rates, e.g. "25600" and "100". This won't work.
>
> For the other error, [...]

As of now, I'm just testing out with `msp` itself and not with ROS support. I wasn't able to try `ros_multiwii` becuase my teammate managed to do it via Python's `pyserial` library. But then that got me concerned because just to work with an FC we'l have to set up some translation mechanism, which is again gonna take our time.

Coder-X15, 2025-03-01:
FYI, I tried `./fcu_motors /dev/ttyACM0 115200` (the serial task was already configured with that frequency) and this is what I got:
```
Connecting to FC...
#FC variant:
 Identifier: BTFL
#Api Version:
 API: 1.45
 Protocol: 0
#FC version:
 Version: 4.4.0
#Board Info:
 Identifier: S7X2
 Version: 0
 OSD support: 2
 Comms bitmask: 55
 Board Name: STM32F7X2
#Build Info:
 Date: Mar  1 2025
 Time: 06:30:43
 Git revision: 4605309
#Status:
 Cycle time: 1974 us
 I2C errors: 0
 Sensors:
    Accelerometer: ON
    Barometer: ON
    Magnetometer: OFF
    GPS: OFF
    Sonar: OFF
 Active Boxes (by ID): 1
#Ident:
 MultiWii Version: <unset>
 MSP Version: <unset>
 Type: UNDEFINED
 Capabilities:
    Bind:   OFF
    DynBal: OFF
    Flap:   OFF
Message marker 65 is not recognised!
Version marker 66 is not recognised!
terminate called after throwing an instance of 'std::runtime_error'
  what():  Cannot get BoxNames!
Aborted (core dumped)
```
When I checked in the Betaflight firmware repo, they did have commands 65 and 116. And yet for some reason the message marker 65 is not being recognised. I shall try and work around this issue somehow. but please do post your thoughts too.

Coder-X15, 2025-03-01:
I set the logging level to debug and retried:
```
~/msp/build$ ./fcu_motors /dev/ttyACM0 115200
Connecting to FC...
sending message - ID 2
sending: 2 | 
packed: 24 4d 3c 0 2 2 
write complete: 6 vs 6
processOneMessage on 0 bytes
buffer returned EOF; reading char directly from port
terminate called after throwing an instance of 'std::system_error'
  what():  read: End of file
Aborted (core dumped)
```

Coder-X15, 2025-03-01:
Uhm wait a minute, it seems like the interface isn't able to read from the FC despite being connected

christianrauch, 2025-03-01:
> When I checked in the Betaflight firmware repo, they did have commands 65 and 116. And yet for some reason the message marker 65 is not being recognised. I shall try and work around this issue somehow. but please do post your thoughts too.

The message marker is not related to the command IDs. It is expecting `$` but read something else, `A` (65) and `B` (66) in this case.

christianrauch, 2025-03-01:
> Uhm wait a minute, it seems like the interface isn't able to read from the FC despite being connected

What do you mean? You showed that it reads:
```
#FC variant:
 Identifier: BTFL
#Api Version:
 API: 1.45
 Protocol: 0
#FC version:
 Version: 4.4.0
#Board Info:
 Identifier: S7X2
 Version: 0
 OSD support: 2
 Comms bitmask: 55
 Board Name: STM32F7X2
#Build Info:
 Date: Mar  1 2025
 Time: 06:30:43
 Git revision: 4605309
```

It is clearly reading some information from the FC.

Coder-X15, 2025-03-01:
> What do you mean? You showed that it reads:
> ```
> #FC variant:
>  Identifier: BTFL
> #Api Version:
>  API: 1.45
>  Protocol: 0
> #FC version:
>  Version: 4.4.0
> #Board Info:
>  Identifier: S7X2
>  Version: 0
>  OSD [...]

What I meant is, after removing the logging level line I added in the main code, I recompiled the whole thing but for some reason, now it isn't displaying the board information as well. Just FYI, we did set the FC to MSP-controlled mode

Coder-X15, 2025-03-01:
> The message marker is not related to the command IDs. It is expecting `$` but read something else, `A` (65) and `B` (66) in this case.

So how should I resolve this?

christianrauch, 2025-03-01:
With all the different working and non-working states, it is actually difficult to follow up on what your actual problem is. If it is just that the `MSP_BOXNAMES` (116) message is not recognised by the FC, then this is probably just because it was removed from the firmware. If you get serial connection issues from time to time, then check if the connection is OK with the Betaflight Configurator.

christianrauch, 2025-03-01:
Also, if you want me to debug the software for a specific type of FC hardware, you have to provide this somehow. I only ever used the `msp` library with the Naze32 and with the original MultiWii on the Arduino Nano.

Coder-X15, 2025-03-01:
> Also, if you want me to debug the software for a specific type of FC hardware, you have to provide this somehow. I only ever used the `msp` library with the Naze32 and with the original MultiWii on the Arduino Nano.

Our hardware list:
1. HolyBro Kakute F7 flight controller 
2. Nvidia Jetson Nano as flight computer
3. Betaflight firmware version: 4.4.0

Coder-X15, 2025-03-01:
> With all the different working and non-working states, it is actually difficult to follow up on what your actual problem is. If it is just that the `MSP_BOXNAMES` (116) message is not recognised by [...]

Unfortunately, the message is in the Betaflight firmware's `msp_protocol.h` file.

Coder-X15, 2025-03-01:
> Also, if you want me to debug the software for a specific type of FC hardware, you have to provide this somehow. I only ever used the `msp` library with the Naze32 and with the original MultiWii on the Arduino Nano.

I'll try and debug the issue that popped in eight after I set the debug level using `FlightController::setLoggingLevel()` function

Coder-X15, 2025-03-01:
> With all the different working and non-working states, it is actually difficult to follow up on what your actual problem is. If it is just that the `MSP_BOXNAMES` (116) message is not recognised by [...]

The serial connection issue popped up after I set the logging level to DEBUG and then I removed the line from the code. But I think even at that time only one message was sent to the FC and the program was terminated with an error.

christianrauch, 2025-03-01:
> Our hardware list:
>
> 1. HolyBro Kakute F7 flight controller
> 2. Nvidia Jetson Nano as flight computer
> 3. Betaflight firmware version: 4.4.0

Well, I meant to provide the hardware. I cannot test on the "HolyBro Kakute F7" since I only have the Naze32 hardware, with an older Betaflight version, to test. If you find another way to reproduce this without the hardware (e.g. in simulation), then let me know.

christianrauch, 2025-03-01:
> The serial connection issue popped up after I set the logging level to DEBUG and then I removed the line from the code. But I think even at that time only one message was sent to the FC and the [...]

I don't see how enabling the debug messages would change the serial communication. Can you give me one complete example (with code) that I can use to reproduce your issue? Maybe just share a branch in your fork.

Coder-X15, 2025-03-01:
> I don't see how enabling the debug messages would change the serial communication. Can you give me one complete example (with code) that I can use to reproduce your issue? Maybe just share a branch [...]

I mean, I don't really know tbh, I'm a student who's into theoretical sides of computing (or even specifically, AI and Data Science), and I have no truck with something that doesn't really come under my purview. I joined the team because I did have some experience in ROS in a previous project. That makes me completely dumbfounded about what just happened. Besides, I couldn't do much more fixing today since I had to do a confession at a nearby church since Lent's gonna start soon, so I'm pretty tired even now. I'll let you know of anything soon. And yeah, thanks for patiently staying along with the dork that I am, I really value your help and would like your help till the time I fix the MSP communication and end up being able to control the drone via the Jetson Nano.

Coder-X15, 2025-03-01:
And tbh I didn't fork it, I just `git clone`d it and did whatever installation steps you said. If it's about the debug thingy, I added `fcu.setLoggingLevel(<some namespace, I forgot everything except that  it starts with 'msp'>::DEBUG)` before `fcu.connect()`

christianrauch, 2025-03-01:
Since I cannot actually debug issues on your hardware, I can only give you pointers where to look. If you are serious about debugging the issue, you need to get more familiar with the code base.

There are essentially three layers to this library. On the lowest layer, closest to the hardware, it is using Asio library for [serial communication](https://think-async.com/Asio/asio-1.30.2/doc/asio/reference/serial_port.html). This sends and receives the raw data. The class [`msp::client::Client`](https://github.com/christianrauch/msp/blob/master/inc/msp/Client.hpp) uses Asio for communication and implements protocol. This is the core of converting the raw data provided by Asio to/from MSP messages. The class [`fcu::FlightController`](https://github.com/christianrauch/msp/blob/master/inc/msp/FlightController.hpp) then just provides some high level functionality on top of the `msp::client::Client`. The `fcu::FlightController` is for convenience. It provides simple functions that wrap the usage of the messages, e.g. it lets you set the control source to MSP or send RC command without the need to know the underlying messages and their structure.

So, if you want to debug the serial communication itself, I would focus on `msp::client::Client` and its examples.

Coder-X15, 2025-03-14:
I'm so sorry for ghosting you for so long, we had decided to switch to APM 2.8 due to the compatibility of the APM stack with `mavros`. We had known about this earlier but with all our equipment we had to go for something pretty small as the one I was talking about. The drone had to weigh less than 2 kg, which is why we had to undergo all this trouble. Thank you for obliging to help, that was very kind of you to have taken your own precious time to help us. I apologise for any loss you may have had due to this loss of time.

closed 2025-03-14 by Coder-X15